ArtificalSUN/Kcalibrator #3: Y size of the bed is not restored/read from the configuration file

dmak, 2021-01-15:
Configuration files reads:
```
bed_size = (220.0, 220.0)
```
however when application `Kcalibrator.exe` starts, Y-size is zero:
[image]

ArtificalSUN, 2021-01-15:
This screenshot is from an old version. Try using the latest one.

dmak, 2021-01-16:
Thanks for update. Unfortunately new v1.0.2 does not start on Windows 7 x64 (v1.0.1 runs just fine):
[image]
Any way to produce debug information (e.g. log) to resolve the issue?

ArtificalSUN, 2021-01-17:
@dmak , are you sure you are having this issue with v 1.0.2? Version 1.0.1 had exactly this behavior when config file was corrupted or from an older version. The only reason for v1.0.2 is to fix this issue. All executables are built in exactly the same way on the same system so I doubt the problem is in your operating system. What is the version string (first line in the output gcode) for the executable that works for you?

dmak, 2021-01-17:
Yes, I have an issue with latest release, downloaded from [here](https://github.com/ArtificalSUN/Kcalibrator/releases/download/v1.0.2/kcalibrator.exe) (MD5 acf20380c409fb475acc1b7282945434). It does not start on Windows 7 and Windows 10.
Previous release, downloaded from [here](https://github.com/ArtificalSUN/Kcalibrator/releases/download/v0.2.0/Kcalibrator.exe) (MD5 a537b19682fc0612ce78973ac3b90c35) launches/runs just fine.

dmak, 2021-02-04:
I've got what was the issue. CFG file created by previous version, causes new version to fail. Namely, if
```
bed_size = (200.0, 200.0)
```
then the program crashes. Otherwise latest version runs just fine.
Anyway, the problem with loading values from saved configuration has been solved.

closed 2021-02-04 by dmak