ArtificalSUN/Kcalibrator #5: Retraction value accept only integer

FedorZa, 2021-02-08:


FedorZa, 2021-02-08:
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "kcalibrator.pyw", line 582, in update_and_create
    currentConfig.updatesettings(top)
  File "kcalibrator.pyw", line 71, in updatesettings
    self.retract = (int(root.retract_length_entry.get()), int(root.retract_speed_entry.get()))
ValueError: invalid literal for int() with base 10: '0.5'
started creategcode
stopped creategcode
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "kcalibrator.pyw", line 582, in update_and_create
    currentConfig.updatesettings(top)
  File "kcalibrator.pyw", line 71, in updatesettings
    self.retract = (int(root.retract_length_entry.get()), int(root.retract_speed_entry.get()))
ValueError: invalid literal for int() with base 10: '0.5'
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "kcalibrator.pyw", line 582, in update_and_create
    currentConfig.updatesettings(top)
  File "kcalibrator.pyw", line 71, in updatesettings
    self.retract = (int(root.retract_length_entry.get()), int(root.retract_speed_entry.get()))
ValueError: invalid literal for int() with base 10: '0.5'

ArtificalSUN, 2021-02-19:
You are using an old version, please use the latest one

closed 2021-06-29 by ArtificalSUN