ArtificalSUN/Kcalibrator #8: Прочистка сопла перед печатью

Dimyc, 2021-06-29:
Еще обно предложение по доработке - печать линии на краю стола перед осносной печатью. Полезна после замены пластика.

; intro line
G1 X3 F1000
G1 Z0.2 F720
G1 Y60 E8.0 F900
G1 Y100 E10.0 F700
G92 E0.0

ArtificalSUN, 2021-06-29:
In principle, the pattern has quite a substantial brim on the first layer, so it should be more than enough to clean the nozzle.
However, I will think about adding an option to clean the nozzle the future.
(I was just too lazy to properly calculate this stuff for delta)

nlef, 2021-07-02:
Maybe custom start & end g-code would be more versatile?

ArtificalSUN, 2021-07-02:
Yes, it is one option. But it might be an overkill just for the purposes of nozzle cleaning. I'm currently considering adding several advanced settings, among them the option to configure the brim. In principle, wider brim should be enough to provide any amout of nozzle cleaning.
However, there is a chance that custom g-code will be added at some point.