ArtificalSUN/Kcalibrator #9: GUI elements overlap each others.

gr8boom, 2021-06-30:
[image]
Standart Kubuntu theme without any special settings.

gr8boom, 2021-06-30:
I changed 88 line of kcalibrator_gui.py
from
        top.resizable(0, 0)
to
        top.resizable(1, 1)
Still looks ugly, but better then nothing.
[image]

ArtificalSUN, 2021-07-01:
Indeed, there is a problem with ttk-based interface that it looks differently on different OS (and different desktop environments). GUI was designed for Windows and I can test it only on Windows, so on other sytems it might look ugly, and it was not designed to be adaptible (bad decision) so resizing it breaks the layout. Perhaps, I cannot into GUI.
It will require some efforts to make it adaptible, but I will consider it.