ArtificalSUN/Kcalibrator #14: Blank default settings on OSX.

dgorodnichy, 2021-07-23:
When I launch Kcalibrator on OSX I don't see any default settings. All select lists are also blank. Because of this, I can't choose values in these fields. When I fill remain settings manually and click on "Generate G code" I don't see any actions. Code is not generated

[image]
 [image]

dgorodnichy, 2021-07-23:
The same behaviour for Windows:
 
[image]

ArtificalSUN, 2021-07-23:
You have to run kcalibrator.pyw, not kcalibrator_gui.py.
I will update the README to make it more clear.

dgorodnichy, 2021-07-23:
Thanks for a quick answer. It helped!
[image]

closed 2021-07-23 by dgorodnichy