ArtificalSUN/Kcalibrator #22: Выбор потока

Gromor322, 2022-01-19:
Хотелось бы иметь возможность задавать поток в программе. Наткнулся на нее в ютюбе, выглядит очень классно, но воспользоваться не смогу, т.к. при калиброванных шагах экструдера и введенном диаметре филамента поток слишком большой, возможно от плотности моего пластика зависит, хотелось бы иметь возможность это учесть при генерации Gcod`a, как в марлиновской утилите.

cydwarf, 2022-01-26:
Я — за!

ArtificalSUN, 2022-09-09:
Thx for the suggestion. Currently you can just mutiply your filament diameter by your desired flow, the result will be the same (as the flow is just a multiplier for extrusion). I.e. if your real filament diameter is 1.73 and you want to set 92% flow, then just set the filament diameter in the program to 1.73*0.92 = 1.592

AlatusAles, 2024-03-07:
Приветствую! Пожалуйста, поправьте если я не прав, возник вопрос. Пример: Заявленный диаметр нити = 1,75 и желаемый поток = 98% (явно нить больше в диаметре от заявленного 1,75, с потоком 100% переэкструзия), 1,75 / 0,98 = 1,786 = реальный диаметр нити. Устанавливать "Filament diameter" = 1,786 ИЛИ по приведенному выше  примеру 1,75 * 0,98 = 1,715 ???

AlatusAles, 2024-11-01:
> Спасибо за предложение. В настоящее время вы можете просто умножить диаметр вашей нити на желаемый поток, результат будет тем же (поток — это всего лишь множитель для экструзии). То есть, если ваш реальный диаметр нити составляет 1,73, и вы хотите установить поток 92%, то просто установите диаметр нити в программе на 1,73*0,92 = 1,592

Формула для указания диаметра прутка в вашем калибраторе будет следующая:
D=2*√((Dn/2)^2/k)

Где:
D - диаметр прутка для калибратора;
Dn - номинальный диаметр прутка (заявленный производителем/измеренный пользователем);
k - коэффициент потока (в формате X.XX) по результатам тестирования при Dn.

[image]

Формула получена на основании выходных файлов G-code вашего калибратора.